alphacharlie/node-ads1x15 #3: install fails because node-gyp rebuild fails

andyadler, 2016-05-18:
This package is just the thing I need for a project I am working on.

When I try to install node-ads1x15 on my raspberry pi 3, the rebuild of node-gyp fails:

pi@smoke-pi:~/work/tools/node-ads1x15 $ npm install node-ads1x15 --save
\
> i2c@0.2.1 install /home/pi/node_modules/node-ads1x15/node_modules/i2c
> node-gyp rebuild

make: Entering directory '/home/pi/node_modules/node-ads1x15/node_modules/i2c/build'
  CXX(target) Release/obj.target/i2c/src/i2c.o
../src/i2c.cc: In function ‘void ReadBlock(const v8::FunctionCallbackInfo<v8::Value>&)’:
../src/i2c.cc:162:47: error: no matching function for call to ‘New(int32_t&)’
   Local<Object> buffer = node::Buffer::New(len);
                                               ^
../src/i2c.cc:162:47: note: candidates are:
In file included from ../src/i2c.cc:2:0:
/home/pi/.node-gyp/4.4.3/include/node/node_buffer.h:28:40: note: v8::MaybeLocal<v8::Object> node::Buffer::New(v8::Isolate*, size_t)
 NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate, size_t length);
                                        ^
/home/pi/.node-gyp/4.4.3/include/node/node_buffer.h:28:40: note:   candidate expects 2 arguments, 1 provided
/home/pi/.node-gyp/4.4.3/include/node/node_buffer.h:31:40: note: v8::MaybeLocal<v8::Object> node::Buffer::New(v8::Isolate*, v8::Local<v8::String>, node::encoding)
 NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                        ^
/home/pi/.node-gyp/4.4.3/include/node/node_buffer.h:31:40: note:   candidate expects 3 arguments, 1 provided
/home/pi/.node-gyp/4.4.3/include/node/node_buffer.h:36:40: note: v8::MaybeLocal<v8::Object> node::Buffer::New(v8::Isolate*, char*, size_t, node::Buffer::FreeCallback, void*)
 NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                        ^
/home/pi/.node-gyp/4.4.3/include/node/node_buffer.h:36:40: note:   candidate expects 5 arguments, 1 provided
/home/pi/.node-gyp/4.4.3/include/node/node_buffer.h:43:40: note: v8::MaybeLocal<v8::Object> node::Buffer::New(v8::Isolate*, char*, size_t)
 NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                        ^
/home/pi/.node-gyp/4.4.3/include/node/node_buffer.h:43:40: note:   candidate expects 3 arguments, 1 provided
i2c.target.mk:86: recipe for target 'Release/obj.target/i2c/src/i2c.o' failed
make: *** [Release/obj.target/i2c/src/i2c.o] Error 1
make: Leaving directory '/home/pi/node_modules/node-ads1x15/node_modules/i2c/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Linux 4.4.9-v7+
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/node_modules/node-ads1x15/node_modules/i2c
gyp ERR! node -v v4.4.3
gyp ERR! node-gyp -v v3.3.1
gyp ERR! not ok 
npm ERR! Linux 4.4.9-v7+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "node-ads1x15" "--save"
npm ERR! node v4.4.3
npm ERR! npm  v2.15.1
npm ERR! code ELIFECYCLE

npm ERR! i2c@0.2.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the i2c@0.2.1 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the i2c package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs i2c
npm ERR! Or if that isn't available, you can get their info via:
npm ERR! 
npm ERR!     npm owner ls i2c
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/work/tools/node-ads1x15/npm-debug.log
pi@smoke-pi:~/work/tools/node-ads1

alphacharlie, 2016-05-18:
Looking at your logs the i2c package is erroring out on build. This appears to be an issue with the i2c package not with the node-ads1x15 library... (See https://github.com/kelly/node-i2c/issues/75 )

closed 2016-05-18 by alphacharlie

alphacharlie, 2016-06-06:
Note - If you are on the Raspberry Pi you can use this version of Node.js as a workaround - http://node-arm.herokuapp.com/node_archive_armhf.deb