alphacharlie/node-ads1x15 #4: error Failed to set address

BJSingh, 2016-07-22:
I tried your library for raspberry pi and ads1115. But when run I get error failed to set address for line wire.setAddress(address); in i2c.coffee file. I have attach code file and error
code file: 
[image]

error file:
[image]

alphacharlie, 2016-07-22:
OK, A couple Questions - 
1. What hardware are you on? (i.e. which Raspberry pi board)
2. Is I2C enabled in raspi-config (and are the modules being loaded at boot?)
3. Does the file **/dev/i2c-1** (or **/dev/i2c-0** ) exist in your filesystem?

BJSingh, 2016-07-23:
I am using Pi3. 
i2c was not enabled, I enabled then I got another error. 
/dev/i2c-1 is showing in file system after i2c enabled.
[image]

BJSingh, 2016-08-05:
Hello alphacharlie, issue has been resolved I am getting following output  when channel 0 connected to GND.
[image]

much close value, I will be testing more and will respond you about results.
Thanks,

alphacharlie, 2016-08-28:
Glad you got it working! I'm now marking this one as closed. ;-)

closed 2016-08-28 by alphacharlie