alphacharlie/node-ads1x15 #19: Received Callback twice

hilschernetpi, 2019-07-22:
We are using the function "readADCDifferential()" (supposing it happens with others too) with our ads1115 chip in our product. To customers we are providing an npm environment as software interface so customer can use your code easily. It is working excellent. 
For us it is obligatory to perform EMC tests to the analog inputs we connected to the ADS chip before we release a final product. During the EMC tests it happens that the I2C bus is disturbed as well and the communication across I2C may get interrupted.

This is why the readADCDifferential() function from time to time returns the error "We've got an Error, Lance Constable Carrot!: Error: Cannot write to device" 
```
self.wire.writeBytes(ADS1015_REG_POINTER_CONFIG, bytes, function(err) {
      if(err)
      {
        self.busy = false;
        callback("We've got an Error, Lance Constable Carrot!: " + err.toString());
      }
    });
```
This is also fine so far ... but now comes the problem: even if the error was reported in the callback (and the write over I2C was not executed) the readADCDifferential() function continues to setup another I2C command "self.wire.readBytes()" in a setTimeout() function and finally we got another callback(null, data);
So with a single call to "readADCDifferential()", we receive two callbacks which is not ok. Also we are wondering which "data" is provided in the second callback if the previous write command failed.
Should be corrected. In case of a write failed a following read should never be executed.

aessig, 2021-02-05:
I have noticed the same issue. @hilschernetpi did you found some fix ?

hilschernetpi, 2021-02-05:
Well not really. We solved it with a work around since we had to come further.  We intercept the problem in the calling user application as you can see in our source code here https://github.com/HilscherAutomation/netPI-nodered-npix-ai/blob/72bafb2c142f985c0a5b933e44f81458d464fb92/node-red-contrib-npix-ai/npixai.js#L61.
If the function readADCDifferential() returns an error as described we set a temporary variable "error" to true. If now our callback function is called again as described in the issue we ignore the call if the variable is true respectively execute the code if the variable is false.